UmerCodez/SensorServer #1: Question or improvement

darld, 2021-11-07:
Can this app start the websocket server automatically during Android phone boot up?

umer0586, 2021-11-08:
> Can this app start the websocket server automatically during Android phone boot up?

No, the Websocket server doesn't start automatically , user has to start it explicitly through the app

opendeliveryrobot, 2021-11-09:
Not sure whether we can root the device to get it start automatically during boot up. For sure, we need the app have function. For my experience, my phone can tether USB once I connect USB cable without need to unlock the device if i root my phone.

umer0586, 2021-11-09:
> Not sure whether we can root the device to get it start automatically during boot up. For sure, we need the app have function. For my experience, my phone can tether USB once I connect USB cable without need to unlock the device if i root my phone.

There is no need to root the device for such purpose, any android app can execute code  when boot up in completed. In such case app needs permission.

    <uses-permission android:name="android.permission.RECEIVE_BOOT_COMPLETED" /> 

And Websocket server must somehow start as background `service` when `android.intent.action.BOOT_COMPLETED` broadcast is received

closed 2021-11-11 by umer0586