UmerCodez/SensorServer #2: Websocket connection failed

yogeshbndm, 2021-11-22:
When i tried to make a client and establish the connection, it shows "connection failed". Can you suggest me the possible reason for this.

umer0586, 2021-11-23:
> When i tried to make a client and establish the connection, it shows "connection failed". Can you suggest me the possible reason for this.

Is there any further information along with `connection failed` ?. Make sure client and server are connected to the same network .

In which language you are writing Websocket client  ? Have you tested the app with any of **Websocket testing tools** available (like http://livepersoninc.github.io/ws-test-page/)

yogeshbndm, 2021-11-25:
issue fixed... the problem was that the client and server were not
connected to the same network...

closed 2021-11-25 by umer0586

Dantrix846, 2022-10-26:
> > When i tried to make a client and establish the connection, it shows "connection failed". Can you suggest me the possible reason for this.
> 
> @yogeshbndm Is there any further information along with `connection failed` ?. Make sure client and server are connected to the same network and client is connecting using valid URL.
> 
> In which language you are writing Websocket client ? Have you tested the app with any of **Websocket testing tools** available (like http://livepersoninc.github.io/ws-test-page/)

i have everything @ the same network, but it still doesnt work even in the ws test page

umer0586, 2022-10-27:
@Dantrix846 make sure you typed url correctly

Dantrix846, 2022-10-27:
it was and is typed correctly. Idk what could be wrong.

umer0586, 2022-10-27:
@Dantrix846 you must be doing something wrong. If your PC and phone are on same wifi network then specifying proper ip and port in websocket client should successfully connect to app.