UmerCodez/SensorServer #3: Hi ; FAILURE: Build failed with an exception.   can you help me. thanks

mvwtest, 2022-01-26:
hello, i use this Action build  your project, but  i got this:
FAILURE: Build failed with an exception.   can you help me. thanks

my workflows file:
https://github.com/mvwtest/SensorServer/blob/main/.github/workflows/SensorServer.yml

and biuld log:
https://github.com/mvwtest/SensorServer/runs/4952176375?check_suite_focus=true

how can i do it? thank you.

mvwtest, 2022-01-26:
Yes,I got it. You are great.thanks.

mvwtest, 2022-01-26:
Asol ,will you contemplate add Bluetooth-uart or usb serial ?

umer0586, 2022-01-26:
> Yes,I got it. You are great.thanks.

@mvwtest Glad to know you solved the issue by yourself. so what was the cause ?

umer0586, 2022-01-26:
> Asol ,will you contemplate add Bluetooth-uart or usb serial ?

USB using ABD has already been added to the app

closed 2022-01-28 by umer0586