UmerCodez/SensorServer #6: Stops running when screen is off

diggit, 2022-07-04:
Hi,
first thing, thank you for this awesome app!

The only issue I have is that app stops providing data when screens shuts down. Connection is not dropped, but data stops flowing.
This issue might be partially similar to #4 .
Power saving is good, but sometimes I need longer recording and keeping screen on for whole time consumes even more power.

umer0586, 2022-07-06:
@diggit 
Actually the underlying sensor stops delivering sensor values when device goes in sleep mode. The only thing a programmer can do is force device to not go into sleep mode using **wake lock** or **partial wake lock** while app runs. After going to through some of [Android official docs](https://source.android.com/devices/sensors/suspend-mode) I came to know that there are two types of sensor wake-up and non-wake-up sensor

>**Non-wake-up sensors**
Non-wake-up sensors are sensors that do not prevent the SoC from going into suspend mode and do not wake the SoC up to report data. In particular, the drivers are not allowed to hold wake-locks. It is the responsibility of applications to keep a partial wake lock should they wish to receive events from non-wake-up sensors while the screen is off. While the SoC is in suspend mode, the sensors must continue to function and generate events, which are put in a hardware FIFO. (See [Batching](https://source.android.com/devices/sensors/batching) for more details.) The events in the FIFO are delivered to the applications when the SoC wakes up. If the FIFO is too small to store all events, the older events are lost; the oldest data is dropped to accommodate the latest data. In the extreme case where the FIFO is nonexistent, all events generated while the SoC is in suspend mode are lost. One exception is the latest event from each on-change sensor: the last event [must be saved ](https://source.android.com/devices/sensors/batching#precautions_to_take_when_batching_non-wake-up_on-change_sensors)outside of the FIFO so it cannot be lost.As soon as the SoC gets out of suspend mode, all events from the FIFO are reported and operations resume as normal.Applications using non-wake-up sensors should either hold a wake lock to ensure the system doesn't go to suspend, unregister from the sensors when they do not need them, or expect to lose events while the SoC is in suspend mode.

>**Wake-up sensors**
In opposition to non-wake-up sensors, wake-up sensors ensure that their data is delivered independently of the state of the SoC. While the SoC is awake, the wake-up sensors behave like non-wake-up-sensors. When the SoC is asleep, wake-up sensors must wake up the SoC to deliver events. They must still let the SoC go into suspend mode, but must also wake it up when an event needs to be reported. That is, the sensor must wake the SoC up and deliver the events before the maximum reporting latency has elapsed or the hardware FIFO gets full. See [Batching](https://source.android.com/devices/sensors/batching) for more details.To ensure the applications have the time to receive the event before the SoC goes back to sleep, the driver must hold a "timeout wake lock" for 200 milliseconds each time an event is being reported. That is, the SoC should not be allowed to go back to sleep in the 200 milliseconds following a wake-up interrupt. This requirement will disappear in a future Android release, and we need this timeout wake lock until then.

For more info Read [isWakeUpSensor()](https://developer.android.com/reference/android/hardware/Sensor#isWakeUpSensor())

Sensor like proximity is a wake-up sensor, it reports data while device is in sleep mode. I also noted that [getDefaultSensor()](https://developer.android.com/reference/android/hardware/SensorManager#getDefaultSensor(int,%20boolean)) has overloaded parameter to acquire wake-up version of sensor if exits

All things said I have to clear my understanding in this regard before jumping into code. This needs some time.

diggit, 2022-07-10:
Today I noticed, that SernsorServer stops providing data when it's running in background/not focused. even when screen is on.

umer0586, 2022-07-11:
> Today I noticed, that SernsorServer stops providing data when it's running in background/not focused. even when screen is on.

It happened on your device only or have you tried the app on other devices too ? I don't had this issue when I test the app on three different devices.

diggit, 2022-07-12:
I didn't have a chance to try it different device. My device is Samsung Galaxy S10e with LineageOS ROM, Android 11.
SensorServer does not stop sending data immediately after it loses focus. It takes approximately 1 minute before it happens.

umer0586, 2022-07-13:
> I didn't have a chance to try it different device. My device is Samsung Galaxy S10e with LineageOS ROM, Android 11. SensorServer does not stop sending data immediately after it loses focus. It takes approximately 1 minute before it happens.

@diggit 
using [livepersoninc.github.io/ws-test-page/](http://livepersoninc.github.io/ws-test-page/) I re-tested the scenario on my Redmi 4A (Android 7.1) with 15 min of accelerometer's data without focus.

diggit, 2022-07-16:
Tested with linked tool and data stops coming after 1 minute since app is moved to background.

umer0586, 2022-07-18:
> Tested with linked tool and data stops coming after 1 minute since app is moved to background.

I am not sure if it's device specific or android OS related issue. It seems your device stop reporting sensor when app is not in focus. Maybe using foreground service would help avoid such problems. 
I don't have time till mid of September for updating the app.

umer0586, 2022-11-15:
> Today I noticed, that SernsorServer stops providing data when it's running in background/not focused. even when screen is on.

App has been updated (Foreground service added). Now app can send sensor data to clients even if its not in focus

closed 2022-12-05 by diggit