UmerCodez/SensorServer #10: Bugsnag

IzzySoft, 2022-11-19:
Last release now includes Bugsnag, which triggered an alert here in my repo (where [your app is listed](https://apt.izzysoft.de/fdroid/index/apk/github.umer0586). Is it configured opt-in, or transmitting data by default? What data is transmitted, and is that transparent to the user (e.g. details are shown and user then can decide to send or not)?

umer0586, 2022-11-19:
No its not configured opt-in, sends data by default  (thrown exceptions). I am not planning to integrate bugsnag in future release, i am just testing there 14 days trial which they have given for free.

IzzySoft, 2022-11-19:
Ah, if it goes away with the next release of course no need to bother. I'll then for now apply the `Tracking` anti-feature (else I get an alert every day) – and once an update removes Bugsnag again I'll receive another alert and will remove the AF again. Thanks for the details!

umer0586, 2022-11-19:
Next release will be uploaded in 8-9 days. Untill then you can mark Tracking anti-feature.

IzzySoft, 2022-11-19:
Done. Will automatically remove it once the update without Bugsnag comes in (my updater will inform me about the dropped tracker). Thanks!

Oh: and feel free to close the issue when it's through :wink:

closed 2022-11-28 by umer0586

IzzySoft, 2022-11-28:
Yay, thanks! As predicted, my updater just informed me to remove the `Tracking` flag – and so I did. Will be gone with tomorrow's sync then.