UmerCodez/SensorServer #17: Not working over WiFi

alexschultze, 2023-01-23:
```
=========== RESTART: \wsocket_client.py ==========
error occurred
[WinError 10061] No connection could be made because the target machine actively refused it
connection close
close code :  None
reason :  None
```

Maybe phone firewall have a reject rule and thus do not work well as servers.

I have tried with two devices (Samsung and Huawei).

umer0586, 2023-01-24:
In my example I used Ip 192.168.0.102 but your android device you may get different IP when you press start button. 
Just make sure that your client script is using correct Ip and Port shown in device

umer0586, 2023-01-26:
@alexschultze still having issue??

alexschultze, 2023-01-26:
I have chosen the correct IP address, as I know devices IP through router, and have ping it with success.
Whats your tested android version?

alexschultze, 2023-01-26:
If this works for you (and others) I suggest closing this issue, as it must be something on my side.

umer0586, 2023-01-26:
Okay but you can try hotspot option. 
I am closing this for now...

closed 2023-01-26 by umer0586