UmerCodez/SensorServer #23: Unable to open app in Android

srbhptl39, 2023-06-04:
I have been trying to open the app [Sensor Server v4.0.0](https://github.com/umer0586/SensorServer/releases/tag/v4.0.0), but it closes immediately. 
Version v3.2.1 works fine

Android Version 13
Oneplus device

umer0586, 2023-06-05:
The next release of the app will include a cash reporting library. Once this is integrated, you will be able to send (crash) logs here.

Have you tested v4.0.0 on some other device with Android Version 13 ?

umer0586, 2023-06-06:
I have tested the app on emulator with android 13 and it doesn't crash on open. Have you tested v4.0.0 on other devices having version 13?

srbhptl39, 2023-06-06:
yes tested in other android 13 version also, it seems to crash. Emulators might not support all sensors which might not be causing it to crash, not sure there.

umer0586, 2023-06-07:
> yes tested in other android 13 version also, it seems to crash. Emulators might not support all sensors which might not be causing it to crash, not sure there.

The app currently does not check for sensors upon opening. The app crashes when opened, indicates that there may be some other issue. I am currently in the process of migrating the code to Kotlin. Once that is complete, I will investigate this issue further.

umer0586, 2023-06-08:
> Can you provide a test version if the integration is complete?

1. Download the installation file [SensorServer-v5.0.0-beta.1.zip](https://github.com/umer0586/SensorServer/files/11683446/SensorServer-v5.0.0-beta.1.zip).
2. After installing, if the SensorServer crashes, it will automatically open your email client.
3. The crash log will be populated in the email body.
4. You have the option to either send the log via email or simply copy and paste it here.

Since GitHub does not allow uploading APK files here, I have provided a ZIP file instead. Please extract APK from ZIP file.

srbhptl39, 2023-06-08:
`{"REPORT_ID":"2edd5d5e-df8c-435a-8b3d-4651ae0dea25","APP_VERSION_CODE":21,"APP_VERSION_NAME":"5.0.0-beta.1","PACKAGE_NAME":"github.umer0586.sensorserver","FILE_PATH":"\/data\/user\/0\/github.umer0586.sensorserver\/files","PHONE_MODEL":"LE2101","BRAND":"OnePlus","PRODUCT":"OnePlus9R_IND","ANDROID_VERSION":"13","BUILD":{"BOARD":"kona","BOOTLOADER":"unknown","BRAND":"OnePlus","CPU_ABI":"arm64-v8a","CPU_ABI2":"","DEVICE":"OnePlus9R","DISPLAY":"LE2101_13.1.0.500(EX01)","FINGERPRINT":"OnePlus\/OnePlus9R_IND\/OnePlus9R:13\/RKQ1.211119.001\/R.103dfa6-1-58fae:user\/release-keys","HARDWARE":"qcom","HOST":"dg02-pool03-kvm246","HW_TIMEOUT_MULTIPLIER":1,"ID":"RKQ1.211119.001","IS_ARC":false,"IS_DEBUGGABLE":false,"IS_EMULATOR":false,"IS_ENG":false,"IS_TREBLE_ENABLED":true,"IS_USER":true,"IS_USERDEBUG":false,"MANUFACTURER":"OnePlus","MODEL":"LE2101","ODM_SKU":"unknown","PERMISSIONS_REVIEW_REQUIRED":true,"PRODUCT":"OnePlus9R_IND","RADIO":"unknown","SKU":"unknown","SOC_MANUFACTURER":"Qualcomm","SOC_MODEL":"SM8250","SUPPORTED_32_BIT_ABIS":["armeabi-v7a","armeabi"],"SUPPORTED_64_BIT_ABIS":["arm64-v8a"],"SUPPORTED_ABIS":["arm64-v8a","armeabi-v7a","armeabi"],"TAGS":"release-keys","TIME":1683259406000,"TYPE":"user","UNKNOWN":"unknown","USER":"root","VERSION":{"ACTIVE_CODENAMES":[],"BASE_OS":"","CODENAME":"REL","DEVICE_INITIAL_SDK_INT":30,"INCREMENTAL":"R.103dfa6-1-58fae","KNOWN_CODENAMES":"{HoneycombMr1, HoneycombMr2, Lollipop, Kitkat, Tiramisu, Gingerbread, Cupcake, IceCreamSandwichMr1, JellyBean, IceCreamSandwich, LollipopMr1, M, N, O, P, Q, R, S, Sv2, Base, NMr1, OMr1, JellyBeanMr1, JellyBeanMr2, Donut, Froyo, GingerbreadMr1, EclairMr1, Honeycomb, Eclair01, KitkatWatch, Base11, Eclair}","MEDIA_PERFORMANCE_CLASS":0,"MIN_SUPPORTED_TARGET_SDK_INT":23,"PREVIEW_SDK_FINGERPRINT":"REL","PREVIEW_SDK_INT":0,"RELEASE":"13","RELEASE_OR_CODENAME":"13","RELEASE_OR_PREVIEW_DISPLAY":"13","RESOURCES_SDK_INT":33,"SDK":"33","SDK_INT":33,"SECURITY_PATCH":"2023-04-05"}},"TOTAL_MEM_SIZE":115823456256,"AVAILABLE_MEM_SIZE":45418471424,"BUILD_CONFIG":{"APPLICATION_ID":"github.umer0586.sensorserver","BUILD_TYPE":"debug","DEBUG":true,"VERSION_CODE":21,"VERSION_NAME":"5.0.0-beta.1"},"CUSTOM_DATA":{},"IS_SILENT":false,"STACK_TRACE":"java.lang.RuntimeException: Unable to start activity ComponentInfo{github.umer0586.sensorserver\/github.umer0586.sensorserver.activities.MainActivity}: java.lang.IllegalStateException: This Activity already has an action bar supplied by the window decor. Do not request Window.FEATURE_SUPPORT_ACTION_BAR and set windowActionBar to false in your theme to use a Toolbar instead.\n\tat android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3876)\n\tat android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:4018)\n\tat android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:111)\n\tat android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)\n\tat android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)\n\tat android.app.ActivityThread$H.handleMessage(ActivityThread.java:2474)\n\tat android.os.Handler.dispatchMessage(Handler.java:106)\n\tat android.os.Looper.loopOnce(Looper.java:240)\n\tat android.os.Looper.loop(Looper.java:351)\n\tat android.app.ActivityThread.main(ActivityThread.java:8399)\n\tat java.lang.reflect.Method.invoke(Native Method)\n\tat com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:584)\n\tat com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1013)\nCaused by: java.lang.IllegalStateException: This Activity already has an action bar supplied by the window decor. Do not request Window.FEATURE_SUPPORT_ACTION_BAR and set windowActionBar to false in your theme to use a Toolbar instead.\n\tat androidx.appcompat.app.AppCompatDelegateImpl.setSupportActionBar(AppCompatDelegateImpl.java:630)\n\tat androidx.appcompat.app.AppCompatActivity.setSupportActionBar(AppCompatActivity.java:185)\n\tat github.umer0586.sensorserver.activities.MainActivity.onCreate(MainActivity.kt:60)\n\tat android.app.Activity.performCreate(Activity.java:8422)\n\tat android.app.Activity.performCreate(Activity.java:8395)\n\tat android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1403)\n\tat android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3849)\n\t... 12 more\n","INITIAL_CONFIGURATION":{"assetsSeq":0,"colorMode":10,"compatScreenHeightDp":680,"compatScreenWidthDp":320,"compatSmallestScreenWidthDp":320,"densityDpi":480,"fontScale":0.9,"fontWeightAdjustment":0,"hardKeyboardHidden":"HARDKEYBOARDHIDDEN_YES","keyboard":"KEYBOARD_NOKEYS","keyboardHidden":"KEYBOARDHIDDEN_NO","locale":"en_GB","mcc":405,"mnc":54,"navigation":"NAVIGATION_NONAV","navigationHidden":"NAVIGATIONHIDDEN_YES","orientation":"ORIENTATION_PORTRAIT","screenHeightDp":765,"screenLayout":"SCREENLAYOUT_SIZE_NORMAL+SCREENLAYOUT_LONG_YES+SCREENLAYOUT_LAYOUTDIR_LTR+SCREENLAYOUT_ROUND_NO","screenWidthDp":360,"seq":11042,"smallestScreenWidthDp":360,"touchscreen":"TOUCHSCREEN_FINGER","uiMode":"UI_MODE_TYPE_NORMAL+UI_MODE_NIGHT_YES","userSetLocale":false,"windowConfiguration":"{ mBounds=Rect(0, 0 - 1080, 2400) mAppBounds=Rect(0, 103 - 1080, 2400) mMaxBounds=Rect(0, 0 - 1080, 2400) mDisplayRotation=ROTATION_0 mWindowingMode=fullscreen mDisplayWindowingMode=fullscreen mActivityType=undefined mAlwaysOnTop=undefined mRotation=ROTATION_0}","mOplusExtraConfiguration":"mThemeChanged= 0, mThemeChangedFlags= 0, mFlip [... cut ...] ionExecutor.executeCallbacks(TransactionExecutor.java:135)\n06-08 20:53:05.519 E\/ACRA (19927): \tat android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)\n06-08 20:53:05.519 E\/ACRA (19927): \tat android.app.ActivityThread$H.handleMessage(ActivityThread.java:2474)\n06-08 20:53:05.519 E\/ACRA (19927): \tat android.os.Handler.dispatchMessage(Handler.java:106)\n06-08 20:53:05.519 E\/ACRA (19927): \tat android.os.Looper.loopOnce(Looper.java:240)\n06-08 20:53:05.519`

srbhptl39, 2023-06-08:
yes, Verified that its working fine with dark mode disabled

umer0586, 2023-06-08:
> yes, Verified that its working fine with dark mode disabled. Expecting a release soon with dark mode too

Yes, next release will support dark mode too

closed 2023-08-01 by umer0586

umer0586, 2023-08-01:
App crash now fixed !

commit e2e9d3b2 (2023-08-01): App crash fix #23
commit 9e5458df (2023-08-01): Merge pull request #29 from umer0586/fix-app-crash

App crash fixed #23