UmerCodez/SensorServer #34: Is it possible to use this over wireguard, listen at 0.0.0.0 instead of wifi/localhost

EntityinArray, 2023-08-28:
Hi, thank you for working on this amazing app!

I have a wireguard network, where my PC is 10.219.135.1 and my smartphone is 10.219.135.2

Your app forces user to have Wi-Fi enabled, but Wi-Fi is absolutely not necessary when using VPN.

I tried checking the Local Host checkbox, but then it only listens to localhost and wg doesn't work either.

Could you add an option to listen on 0.0.0.0, so that users can use VPNs?

umer0586, 2024-02-17:
Ability to listen on 0.0.0.0 has been added in v5.2.0

EntityinArray, 2024-02-17:
> Ability to listen on 0.0.0.0 has been added in v5.2.0

When is it dropping in F-Droid?

umer0586, 2024-02-18:
> When is it dropping in F-Droid?

The F-Droid build server typically exhibits a build, signing, and publication process that requires 3 to 4 days to complete for new updates.

umer0586, 2024-02-22:
v5.2.0 is now available on F-Droid

closed 2024-02-22 by EntityinArray