UmerCodez/SensorServer #56: Server can not be stopped from within the App

matthias-buchmeier, 2024-05-03:
Phone: Samsung Galaxy S22; Android 14; Kernel 5.10.177; One UI version 6.0.

Bug: The server does not stop after activating the stop button on the main screen of the App. The App and server even keep running after closing it. Only "force stop option from Settings/Applications works.

umer0586, 2024-05-05:
- Does this behavior always occur on your device? In other words, have you noticed this happening every time you try to stop the server?
- Are you able to reproduce the same issue on any other Android devices?
- Have you tried clearing the app's cache and storage?

Sometimes clearing an app's cache and storage can resolve temporary glitches.

matthias-buchmeier, 2024-05-06:
The behavior is always reproduced on this device. I never observed it on other devices. Clearing the cache does not help.

umer0586, 2024-05-09:
It sounds like the issue might be specific to your device. 
Have you tried stopping the server from the notification bar?

wijaflo, 2024-06-03:
I'm also unable to stop the connection from within the app. 

Galaxy A52
One UI-Version 6.0
Android Version 14
Kernel 4.14.190

umer0586, 2024-06-04:
@wijaflo 
I don't know why the app is showing such behavior on your and @matthias-buchmeier's device

itamark-targa, 2024-06-07:
Same problem, Galaxy A52s, Android 14
Server still running even if I force close the app,

umer0586, 2024-06-09:
@itamark-targa Noted !

umer0586, 2024-07-23:
This issue has been resolved in [v6.1.1](https://github.com/umer0586/SensorServer/releases/tag/v6.1.1). Please check if the problem still persists on your device.

itamark-targa, 2024-07-25:
> This issue has been resolved in [v6.1.1](https://github.com/umer0586/SensorServer/releases/tag/v6.1.1). Please check if the problem still persists on your device.

Thanks, now the server can be stopped :)
Another issue (not sure if it's related), is that if I close the app (by "up swiping" it) and re-open, pressing the "start" button gives a "port already in use" message. It can be resolved by force-stopping the app.

umer0586, 2024-07-26:
> Another issue (not sure if it's related), is that if I close the app (by "up swiping" it) and re-open, pressing the "start" button gives a "port already in use" message. It can be resolved by force-stopping the app.

I am unable to reproduce this issue on an Android 14 emulator using the following steps:
1. Open the app.
2. Start the server.
3. Close the app by swiping up from the apps list while the server is running.
4. Reopen the app.
5. A message appears: "Service running."
6. Stop the server.
7. Start the server again.
8. A message appears: "Server started."

closed 2024-07-30 by umer0586