UmerCodez/SensorServer #62: Still cannot get GPS location

FitzWang, 2024-08-02:
Hi, I followed the instruction on this link, https://github.com/umer0586/SensorServer/wiki/Getting-Data-From-GPS.
But the python script (both two scripts) can still not get the location.
The python only output 'connected!' but no information even I move the phone from place to place.

The information of my phone is:
Galaxy S21 5G
one UI version: 6.1
android version: 14

Note that I use USB to connect the device with PC, all other sensors work perfectly.

FitzWang, 2024-08-03:
> > The python only output 'connected!' but no information even I move the phone from place to place.
> 
> GPS typically doesn't work well indoors due to signal interference. Are you moving your phone in a room ?

Oh yes, I tested it in a room. I will try outdoors again.
I am not expert of Android, but I'm just wondering, why I can get location through the map APP in a room?

umer0586, 2024-08-03:
> why I can get location through the map APP in a room?

The reason you can get a location through a map app even when indoors is that map applications often use multiple providers to determine your location. Many map apps, including those on Android, utilize the Google Fused Location API, which is a part of Google Play Services. This API can combine data from GPS, Wi-Fi, cell towers, and other sensors to provide an accurate location, even in situations where GPS signals might be weak or unavailable, such as inside buildings.

see https://stackoverflow.com/a/6775456/9193164 for more detail

FitzWang, 2024-08-04:
> The reason you can get a location through a map app even when indoors is that map applications often use multiple providers to determine your location. Many map apps, including those on Android, [...]

Thanks for the explanation. Here are another 2 questions:
1. why can we use "FUSED_PROVIDER" then it won't wait just for the GPS availability.
2. I checked the LocationManager page from android documents, and I used getCurrentLocation as request message (try to trigger the location calculation) in my python script. However, after connection, it outputs error.

[image]

umer0586, 2024-08-04:
> why can we use "FUSED_PROVIDER" then it won't wait just for the GPS availability.

The FUSED_PROVIDER is only available in API level 31 (Android 12 or later), which means it may not be supported on older devices. In contrast, the GPS_PROVIDER is available on almost every Android device, including older ones.

> I checked the LocationManager page from android documents, and I used getCurrentLocation as request message (try to trigger the location calculation) in my python script. However, after connection, it outputs error.

The app currently only responds to the "getLastKnownLocation" message and does not support the "getCurrentLocation" method, which was introduced in Android 13. This feature may be added in future versions of the app.

https://github.com/umer0586/SensorServer/blob/bcf900e113580b150de6a67435ed0827e4cf0f66/app/src/main/java/github/umer0586/sensorserver/websocketserver/SensorWebSocketServer.kt#L380-L390

FitzWang, 2024-08-05:
OK, I changed another phone and get the location. I think the reason that why the previous one cannot get location is that
1 . it's tested in room.
2. it has no sim card.
I think this issue can be closed.

closed 2024-08-05 by FitzWang