UmerCodez/SensorServer #67: Stream on device boot

Manouel96, 2024-09-30:
Hi @umer0586,

Thanks for the nice job.
Can we add an option to start the application and run the server automatically on device boot?
This will help a lot!

Regards,
Manouel

umer0586, 2024-10-01:
> Can we add an option to start the application and run the server automatically on device boot?

While it's possible to implement, could you clarify the use case for this functionality?

Manouel96, 2024-10-01:
I am trying to put the phone somewhere that is hard to reach and collect data from sensors, while I have to turn off and turn on the phone sometimes with a wireless controlled setup I have made, it will be great that when the phone turns on, i don't need to open up the application and start the server manually, this give more reliability for the project, also I was thinking that this can be used when you don't need to have a screen on the phone, I mean you have only the motherboard of the phone working reliably and collecting data.

umer0586, 2024-10-01:
That sounds like a useful scenario, and I would definitely like to add this feature to the app. However, it will take some time and can't be implemented right away. Currently, I'm working on another sensor streaming app that sends sensor data over UDP, and I believe it would be easier to integrate this functionality into that project.

Manouel96, 2024-10-01:
That's wonderful, and is that project already available? It would be nice if the option be available in this app also, because the app and the websocket is stable enough to let it run for hours.

Thanks for your response, and for your great work!

umer0586, 2024-10-02:
That project isn't available yet, but I'll definitely consider adding this feature to the current app as well. Once I have some time, I’ll work on implementing it.

Manouel96, 2024-10-02:
That's really nice! I will be waiting for it!

umer0586, 2024-10-08:
The new project is now available, but it does not yet include the stream-on-boot functionality.

umer0586, 2024-10-11:
@Manouel96  
Is your wireless controller setup capable of enabling Wi-Fi or a hotspot? Starting with Android 10 or above, apps are not allowed to enable Wi-Fi programmatically due to system restrictions. However, devices running Android 9 and below can still perform this action.

Manouel96, 2024-10-11:
Hi @umer0586 , I am using Android 7.0, so i think this feature should be possible.
Please correct me if i am wrong.
Thanks.

umer0586, 2024-10-11:
So this functionality will work only for older device

umer0586, 2024-10-14:
Functionality now added in [SensaGram](https://github.com/umer0586/SensaGram) app